HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #2: SSD1306 Only?

perezjosue, 2019-10-29:
Is it possible to use a different (bigger) display? 

I would like to get a bigger display to view the data from further distance.

HomeKidd, 2019-10-29:
The HomeKit code needs more resources so the firmware is limited to **SSD1306 displays** with **i2c bus* and maximum resolution of **128x64 pixels**. If the bigger display using i2c bus (and also SSD1306 based) and the resolution is exactly the same 128x64px it will work👌

closed 2019-10-29 by HomeKidd

perezjosue, 2019-10-29:
> The HomeKit code needs more resources so the firmware is limited to **SSD1306 displays** with *_i2c bus_ and maximum resolution of **128x64 pixels**. If the bigger display using i2c bus (and also SSD1306 based) and the resolution is exactly the same 128x64px it will work👌

Thank you for the feedback and for creating this! 

I will place my SSD1306 order tonight but will try this without the screen in the meantime.