HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #3: Battery operated?

perezjosue, 2019-10-29:
Have you tested this with any sort of battery?

HomeKidd, 2019-10-29:
This firmware is using HAP (HomeKit accessory protocol) over IP. It needs **constant Wi-Fi connection** so battery is not an option because the accessories will show “No respond” within Home app when they are in deepsleep. For battery powered project i’m started working on Zigbee (slave) devices when the ESP8266 acts as a HomeKit bridge💪🤓

closed 2019-10-29 by HomeKidd

perezjosue, 2019-10-29:
> This firmware is using HAP (HomeKit accessory protocol) over IP. It needs **constant Wi-Fi connection** so battery is not an option because the accessories will show “No respond” within Home app [...]

I cant wait to see that project! Thanks for your feedback!