HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #7: Device stop responding after few hours

CollaVinilica, 2019-11-28:
Hi, using the latest firmware I get “Not responding” in the Home App every few hours. I tried a different ESP8266, same result. 
I can only get it to appear again by resetting the device holding GPIO 0 to ground for 10 seconds but after few hours I get the same problem over and over again. 
If I use a serial monitor I can see the ESP8266 correctly getting the temperature and the humidity and no error whatsoever. 
No HomeKit devices have this behavior. Any ideas? Thanks a lot.

HomeKidd, 2019-11-29:
Hi!

I have no issues with this version of my firmware, 4 ESPs running now since last week and never has No Response issues. Also what changed since the last firmware (2.0.4) is only one OLED image and the screen on/off response so these can't affect nothing.

But [esp-homekit](https://github.com/maximkulkin/esp-homekit-demo/issues?utf8=✓&q=no+response) has tons of these issues, and my firmware is based on this.
Also what may can affect the esp-s is the wifi network itself. I have 3 AP at home and the wifi based _original_ HomeKit accessories sometimes not reachable when i was connected to e.g: AP 1 and the accessory to AP 2. These happens randomly sometimes. Also did you tried to reboot the ESP instead of reseting?

CollaVinilica, 2019-11-29:
Hi, I'm not saying that your code is faulty, I'm just asking if you experienced similar behaviour. I didn't mean to be rude. 
I did tried to reboot the ESP, but never showed up in the Home app again, even tho it connected to the AP without any trouble. 
So you think that might be due to the esp-homekit library? 
I flashed the latest release 2.0.5, I'll see how that will work out. 
Just one last thing: what about the OLED screen? If I didn't get this wrong, the display is supposed to turn off after some time but it never does, randomly I can see your logo but it never shuts down. If I push the button connected between GPIO0 and GND in the serial monitor I get "Unknown button press" or something like that (I don't have it connected to my computer right know). Am I doing something wrong? Thank a LOT for your replay and your great project!

HomeKidd, 2019-11-29:
Yes thats a common problem with esp-homekit beacuse esp8266 doesnt have the correct MDNS library that supports Apple/HomeKit, i've talked with the developer of esp-homekit and Maxim said that i have to update my esp-homekit this maybe fixes the issue. I'll release a new update (as a pre-release) for testing! 

The OLED display works correctly, see [release notes for v2.0.4](https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/releases/tag/2.0.4) 😄 The Logo should appear every ~1-2 minute so that part works correctly too!
"Unknown button press" is seems to be a hardware issue (maybe the button itself). Actually i've never seen this problem with nearly 30+ devices 😳
Button can support single, double (not implemented) and long press. Actually every press on the button which is  < 10 sec should be recognised as 'single press' and it has to work 😄 
In case you have a NodeMCU, the flash button (as i've know) is connected to GPIO0 and GND. 
How you connected the button? Could you post an image?

CollaVinilica, 2019-11-30:
Hi there, I flashed the release 2.0.5 and it seems to be working great so far. 
Except that OLED thing, it stays on regardless of what I do. I’m using a nodemcu  v2 instead of a D1 mini like in the hardware setup page, maybe is that the cause? 
The button is a simple PCB button hooked between GND and GPIO0. I even tried a pull-up (4.7k) but it didn’t seem to make a difference.  Not event with the built in “flash” button, as you suggested. 
That’s not a huge problem, it’s kind of nice to be able to just take a look at the display anytime I swing by the sensor. Maybe I’ll even design a board for it 😁
Anyway.. I didn’t have that “not responding” thing yet, so that seems solved!

HomeKidd, 2019-11-30:
Great! I'll close the issue.

Yes, maybe thats the cause of the button problem!
I've checked the schematic of NodeMCU and GPIO0 is connected to the button and GND as it should be, but also its connected to the RST button and the USB driver chip.
On Wemos side, D1 mini has GPIO0 connected to GND too and the USB driver chip (for automatic flashing, you dont need to push any button the USB chip automatically pulls GPIO0 to GND) and the Wemos dont have any button attached to GPIO0 😄 so i think this is the main issue!

Schematic of the NodeMCU:
[image]

closed 2020-04-02 by HomeKidd