HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #8: didn't work for me

ta4ukoma, 2019-12-03:
Hi! I'm trying to copy you project with esp8266 NodeMCU. Did all as you wrot in wiki, but ...
[image]
Flash is allright, but ssid did't see.
Can you help me ?
Sorry for my English, just yet study.

HomeKidd, 2019-12-03:
Hi!

Actually error message shows that "Please use the equivalent size '4MB'.
So your correct flash parameters:

```
esptool.py -p /dev/<your_ESPPort> --baud 115200 write_flash -fs **4MB** -fm dout -ff 40m \
0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin
```

Also read #5 Issue, this maybe help!

ta4ukoma, 2019-12-03:
Thx a lot for helping =) your advice helped.
But I has a new problem. Wifi ssid again didn't visible and I connected to console esp on baudrate 115200 and saw this:
[image]
Without bme280 and lcd will not work ? I'm connected lcd and bme280 at the same pin like on you pics, but I have nodemcu against you wemos.

HomeKidd, 2019-12-03:
Yes without the BME280 and OLED the Wifi settings won't come up.

Make sure you have the right connections:

**BME 280 pins:**
`SCL =  GPIO5  - D1 on NodeMCU`
`SDA =  GPIO4 - D2 on NodeMCU`

**OLED Screen pins:**

`SCL =  GPIO14  - D5 on NodeMCU`
`SDA =  GPIO13 - D7 on NodeMCU`

Also make sure to have both the sensor and screen connected to 3.3V! Read Issue #6 why!

ta4ukoma, 2019-12-03:
again, thx a lot! It turns out I do not have such a sensor and screen that are required ) buy them the other day and get back to you

closed 2019-12-03 by HomeKidd

ta4ukoma, 2019-12-03:
WoW! It's working amaizing! 
But can I take pressure in mm against hPa ?

HomeKidd, 2019-12-03:
@ta4ukoma Great!

Sorry but only hPa and inHg are available (HomeKit only can switch between these two)!