HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #10: [Client 3] Failed to compute SRP shared secret (code -2)

vvlang, 2020-02-16:
[image]

HomeKidd, 2020-02-16:
Hi! Thats a common issue due to high memory usage in esp-homekit, simply just restart the esp and try again as i’ve mentioned in the [Setup Wiki page](https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/wiki/Setup#homekit-setup)! ✌️

vvlang, 2020-02-16:
Resetting device to factory defaults not solved，Version 1.0.4 is normal！

HomeKidd, 2020-02-16:
Not resetting, simply just restarting aka unplug the power and repower it✌️

vvlang, 2020-02-16:
You are really a talented！！！

HomeKidd, 2020-02-16:
Is it working now?

vvlang, 2020-02-16:
Yes！！
[image]

closed 2020-02-16 by HomeKidd