HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #12: Newer version without OLED-display?

luc-ass, 2020-05-02:
Hey there, first of all thank you so much for creating this! Is there a way to run the newer versions without an OLED-display? So far I haven't had any luck... Version 1.0.4. works perfectly.

Greetings, Lucas

HomeKidd, 2020-05-02:
Hi!
The non-OLED code will not be maintained in the future because most people needs the OLED version. Also not many things are changed since the non-OLED version, only the OLED code changed and some tweaks added ( e.g.: to show the User Manual via Eve app). These features are not must-have changes! In the future ( when i figure out the Data History) I'll release a 1.x.x version for those who want this without screen 😄

closed 2020-05-02 by HomeKidd

33-01, 2020-05-02:
No, i have Oled)
But I wanted to put this device on the roof of the house, adding a battery
and a solar panel😇

HomeKidd, 2020-05-02:
@33-01 

This code is not suitable for battery operation, read more about it in #3 Issue 😄 Even with a big battery i think until morning the batteries will die. 
Also this issue was opened for non-OLED display, if you have an OLED you're good to go because the OLED code is still maintained...😉

33-01, 2020-05-02:
сб, 2 мая 2020 г. в 16:21, HomeKid <notifications@github.com>:

> @33-01 <https://github.com/33-01>
>
> This code is not suitable for battery operation, read more about it in #3
> <https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/issues/3>
> Issue 😄  Thanks 😊
>