HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #13: Optional second sensor?

luc-ass, 2020-05-02:
Hello again, I just wondered wether it would be possible to add a secondary (optional) sensor? My use case would be:
1) Fridge/freezer temperatures
2) Inside/outside temperatures

I would be more than happy to help out if something is required!

Greetings, Lucas

HomeKidd, 2020-05-02:
Hi! 

Second sensor is possible, but since these sensors using i2c bus for communication its not possible to make wires longer than 15-20cm. Also the BME280 needs hardware modification (soldering) to select the I2C bus address because two sensor with the same address will cause problems. Also adding two sensors for one esp8266 is not requested by anyone. So for that purpose ( inside/outside temp) you have to use two separate modules!

closed 2020-05-02 by HomeKidd

33-01, 2020-05-02:
1 Fridge/freezer temperatures)