HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #14: 2.1.4 - can not connect to my Wifi

denis-mrd, 2020-05-04:
After flashing my NodeMCU V3:
```
esptool.py -p /dev/cu.SLAB_USBtoUART erase_flash
esptool.py -p /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fs 4MB -fm dout -ff 40m 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 main.bin
```
I connected sensor and display and powered up with USB adapter. Access point HomeKidd requie a password when I connect form iPhone, but MacBook can connect without pass - is it normal? Auto openning website not working. I opened 192.168.4.1 manually, choose my wifi and paste the password. After 20 min waiting for a connection device to my network - I give up. After some reboots and reflashing - nothing changed

HomeKidd, 2020-05-05:
Hi!

The password bug is a well known issue in [esp-homekit](https://github.com/maximkulkin/esp-homekit) but it is patched now. Also it doesnt affect anything else since the captive portal is just for adding the device to your network. Also since the latest iOS and MacOS update (13.4.x) how the captive portals are working changed and sometimes it takes a long time to render the web UI for the captive portal. More about that [here](https://github.com/HomeKidd/Homekit-WS2812B-controller/issues/12).  I'll release the the patched version soon!

Also I've updated the [Wiki page](https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/wiki/Build-Instructions#installation-with-ota) because there was a wrong parameter: the flash mode for flashing the ESP8266 is not Dout, the right setting is **DIO** ( or QIO... depending on your module). Since the flash used for storing the wifi/homekit settings this is a key point. With this flash setting it should work now! I've running 2.1.4 since March on 3 devices with no issue 😄 

If the problem with the changed flash settings is gone, please close the issue!

denis-mrd, 2020-05-06:
Thank you, it helps!)

closed 2020-05-06 by denis-mrd