HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #15: 1,3 OLED need to calibrate

denis-mrd, 2020-05-04:
I tried firmwares v2.0.5, 2.1.3, 2.1.4 but they not working with my 1,3 OLED display from diymore (aliexpress). 
When device is powered up - there is a lines or somesing like  on the screen. In few bottom rows image changes. I think, it just need to move up more than a half of screen.

HomeKidd, 2020-05-05:
Hi!
First of all upload a picture from the Oled display. It seems to be a hardware issue to me or not supported OLED display. Only SSD1306 display supported with 128x64px🤓

denis-mrd, 2020-05-06:
I bought exactly what you recommended in previous wiki [commit](https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/wiki/Hardware-setup/_compare/cea961fc7e42d8ac4e723b29ac434c2b7be9bd94?diff=unified#diff-87deb8d4954ce7a30af04915cfee62aeL11): [1.3” SSD1306 OLED display](http://s.click.aliexpress.com/e/_sq6JHh)<br/>

[image]

HomeKidd, 2020-05-06:
I've tested my SSH1106 display _(with the Air Quality code which shares the same source code for display)_ and mine working well 😳 
Still seems to a faulty display for me! 
SSH1106 is also compatible with this code, but i've updated the link because the Aliexpress seller seems to be changed the item from SSD1306 to SSH....

[image]

denis-mrd, 2020-05-25:
Nope)

closed 2020-05-25 by denis-mrd

HomeKidd, 2020-11-20:
@denis-mrd  I have some time and figured out a solution for the issue, now SH1106 screens are also supported, but currently in my [Air Quality Sensor code](https://github.com/HomeKidd/ESP8266-HomeKit-Air-Quality-Sensor-Elgato-Eve-Room/releases/tag/2.1.0) 😄 Update will be coming soon for this sensor too! 😄

djashjones, 2021-01-20:
I'm having a very similar issue, also in homekit & EVE it's telling me unreachable, even thou it has a good wifi signal.