HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #18: devices not found

tibur79, 2020-08-17:
day after several attempts with both main.bin and otaboot I can't connect the device to homekit I tried various parameters like 0x0000 0x1000 0x2000 dio qio both with d1mini and
AZDelivery NodeMCU ESP8266 ESP-12F WiFi Lua Lolin V3 the problem remains with ota from the iphone settings makes you connect to lcm ... but then nothing even after 20 minutes of waiting does nothing with main immediately starts flashing the devices but from settings do not see what could be iphone with ios 13.6.1 thanks

HomeKidd, 2020-08-17:
Hi!

Sadly the OTA part is not my code, it was made by [HomeACcessoryKid](https://github.com/HomeACcessoryKid)! OTA downloading process is very buggy, takes at least 7 minutes to download the firmware, also it very often fails downloading it.

Without OTA it should work! Simply just erase the flash and upload the main.bin instead of otaboot.bin as mentioned in the [Wiki](https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/wiki/Build-Instructions#installation-without-ota)!
If you want to update the firmware you can simply flash the newest main.bin **without erasing the flash**. Every setting should be kept 😄

>  immediately starts flashing the devices

Did you connected the OLED and BME280 before? Without any connected sensor it wont start properly...

Also others reported similar issues that has been solved. #14 , #8 , #5

tibur79, 2020-08-17:
thanks for the help but i'm trying them all and for now nothing even with the monitor r sensor connected

HomeKidd, 2020-08-17:
Could you send me logs using serial monitor what is happening when you start the device?

tibur79, 2020-08-17:
[image]
[image]

HomeKidd, 2020-08-17:
I'm interested in the LOG files from the ESP itself (using Arduino's serial monitor)
[image]

Or using [MacOS Terminal](https://apple.stackexchange.com/questions/32834/is-there-an-os-x-terminal-program-that-can-access-serial-ports)!

tibur79, 2020-08-17:
how is the command

HomeKidd, 2020-08-17:
> Or using [MacOS Terminal](https://apple.stackexchange.com/questions/32834/is-there-an-os-x-terminal-program-that-can-access-serial-ports)!

tibur79, 2020-08-18:
I send the command but it says screen finished and does nothing sorry

HomeKidd, 2020-08-18:
You can run `ls /dev/{tty,cu}.*` in Terminal and then `screen /dev/tty.usbserial-A800e3ZL 115200`
It worked for me!

Other ( and best way) is to use [Arduino](https://arduino-esp8266.readthedocs.io/en/latest/installing.html) to capture serial logs.

tibur79, 2020-08-18:
broken monitor

tibur79, 2020-08-18:
monitor rotto

closed 2020-08-18 by tibur79