HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #20: Question: Which BME280

Caleb79, 2020-11-24:
Hi,
I'm now starting to order all parts for the PCB and I'm wondering which BME280 I have to take, 3.3V or 5V.
I think it is 3.3V but all parts I searched for have 5 pins instead of 4 as it is shown in your drawing.

Can you please help me out? Maybe add this information also in the parts list.

Thanks.

HomeKidd, 2020-11-24:
Hi!
The [Wiki](https://github.com/HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve/wiki) has links for the correct BME280! Both on the main page and on the hardware config page.
 
BTW both should work _(3.3V/5V)_, but for the custom PCB only the 4 pin version can be used and this is the recommended also😄

closed 2020-11-24 by HomeKidd

Caleb79, 2020-11-24:
sorry, but not all of the links are working:
the button for example isn't.