HomeKidd/ESP8266-HomeKit-Weather-Station-BME280-Elgato-Eve #22: 3D printable case

djashjones, 2021-01-10:
My PCB's are coming in a couple of weeks.

Any idea on when the STL's files will be available so I can print it?

Really looking forward to building this 🥰

Caleb79, 2021-02-16:
Same here.
Even if it's still under development. Maybe together we can finish on that.

IonutMindrescu, 2021-03-28:
Hi there, any updates for the 3D case? :D

Caleb79, 2021-12-09:
Asking again for an update..

domi256, 2022-07-08:
Hey, maybe something like this. 

https://www.thingiverse.com/thing:2821217

I use this since a while an works pretty good.

IonutMindrescu, 2022-07-08:
Hey @domi256 , thanks for the file. The subject was about the 3D case for the custom PCB but I think I will go with the traditional Wemos D1. It looks like the owner of this project doesn't care anymore about his Git repo.

domi256, 2022-07-08:
oh okay alright. 

sorry didn't knew that.